evilsocket/pwnagotchi #1145: [BUG]

NachiThePanda, 2023-03-09:
**Describe the bug**
I have made a fresh install of the Pwnagotchi ISO on a Raspberry Pi Zero WH and have not been able to connect to the pwnagotchi. Windows 11 does not detect the pwnagotchi and installing RDNIS drivers does not resolve the issue 

**To Reproduce**
Steps to reproduce the behavior:
1. Plug In pwnagotchi
2. Check devices/networks

**Expected behavior**
I expected the Pwnagotchi to show up in devices as an RDNIS gadget, especially since I have drivers installed for RDNIS that have worked on Windows 10.

**Environment (please complete the following information):**
 - Pwnagotchi version
 1.5.5
 - OS version
 Windows 11 22H2
 - Any additional hardware used
 High speed Micro USB cable (used 2 different ones for troubleshooting)

**Additional context**
Someone mentioned they have the same issue in #975 with the same OS (Windows 11)

Osbareat, 2023-09-03:
same here windows 11 and 10

azrael-tzel, 2023-09-27:
> Duo-rndis
> 
> 15:30:
> 
> > Works for me after installing the drivers. On windows 10 and 11.
> >
> > How?

I did, that worked fine. But how do i enter the pwnagotchi? It says it recognize it as a ethernet device. But i cant do anything with it.

Probably i just miss something out, but what?

azrael-tzel, 2023-09-27:
I did on my macbook. 

Couldent do it for some reason on the windows.

Ip is 10.0.0.2 just the regular

azrael-tzel, 2023-09-27:
Ok i will google some tutorial for exactly that. Got any recommendations? Or tips? 

Its been a long time since i even worked with windows 😅

azrael-tzel, 2023-09-27:
I managed to set it in windows.

Now i will try to connect to it......

azrael-tzel, 2023-09-27:
Its aliveee 👾 

Just had to play around with the last numbers of the ip in terminal 🤷 pretty weird.

But it works 👍🏻

joelis10, 2024-01-13:
> I managed to set it in windows.
> 
> Now i will try to connect to it......

how did you manage to set it in windows?

DittmannCz, 2024-12-14:
which rndis driver did you use? mine crashes win11 into BSOD