evilsocket/pwnagotchi #1147: [BUG] Brand new Zero WH and Waveshare 2.1 e-paper display not working

CuttlefishJones, 2023-03-16:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Plugin in display
2. Power on Pi - either data or power
3. ...

**Expected behavior**
The display is expected to show the interface

**Environment (please complete the following information):**
 - Pwnagotchi version - 1.5.5
 - OS version -  4.19.127-Re4son+ #1 Sun Feb 21 21:13:51 AEDT 2021 armv6l GNU/Linux
 - Type of hardware - Pi Zero WH
 - Any additional hardware used - no

**Additional context**

When running clear the following error occurs

pi@albert:/ $ sudo pwnagotchi --clear
[2023-03-16 18:07:40,709] [INFO] Logtail plugin loaded.
[2023-03-16 18:07:40,711] [INFO] [update] plugin loaded.
[2023-03-16 18:07:40,713] [INFO] memtemp plugin loaded.
[2023-03-16 18:07:40,719] [INFO] gps plugin loaded for /dev/ttyUSB0
[2023-03-16 18:07:40,718] [INFO] webcfg: Plugin loaded.
[2023-03-16 18:07:40,716] [INFO] grid plugin loaded.
[2023-03-16 18:07:40,729] [INFO] Session-stats plugin loaded.
[2023-03-16 18:07:40,727] [INFO] [led] plugin loaded for /sys/class/leds/led0/brightness
[2023-03-16 18:07:40,892] [INFO] initializing waveshare v2 display
Unhandled exception in thread started by <function locked_cb at 0xb65ab270>
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/plugins/__init__.py", line 85, in locked_cb
    cb(*args, *kwargs)
  File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/plugins/default/ups_lite.py", line 83, in on_ui_update
    if capacity <= self.options['shutdown']:
[2023-03-16 18:07:40,924] [WARNING] non fatal error while updating view: dictionary changed size during iteration
KeyError: 'shutdown'
Unhandled exception in thread started by <function locked_cb at 0xb65ab270>
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/plugins/__init__.py", line 85, in locked_cb
    cb(*args, *kwargs)
  File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/plugins/default/ups_lite.py", line 83, in on_ui_update
    if capacity <= self.options['shutdown']:
KeyError: 'shutdown'
^CTraceback (most recent call last):
  File "/usr/local/bin/pwnagotchi", line 149, in <module>
    display = Display(config=config, state={'name': '%s>' % pwnagotchi.name()})
  File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/ui/display.py", line 18, in __init__
    self.init_display()
  File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/ui/display.py", line 78, in init_display
    self._implementation.initialize()
  File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/ui/hw/waveshare2.py", line 62, in initialize
    self._display.init(self._display.PART_UPDATE)
  File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/ui/hw/libs/waveshare/v2/waveshare.py", line 173, in init
    self.reset()
  File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/ui/hw/libs/waveshare/v2/waveshare.py", line 149, in reset
    delay_ms(200)
  File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/ui/hw/libs/waveshare/v2/waveshare.py", line 72, in delay_ms
    time.sleep(delaytime / 1000.0)
KeyboardInterrupt

Cardy165, 2023-03-18:
Did you buy your waveshare 2.1 hat recently ?

If so your likely hitting the issue I came across which is its actually a v3 rather than a v2 hat. 

There is already a code fix just not yet in a release. 

[https://github.com/evilsocket/pwnagotchi/pull/1069](https://github.com/evilsocket/pwnagotchi/pull/1069)

I manually made the changes to the 6 files relative to the copy in: /usr/local/lib/python3.7/dist-packages/pwnagotchi on the pi. 

Updated my config to waveshare_3 and it now works.

kontrakote, 2023-08-23:
`File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/plugins/default/ups_lite.py", line 83, in on_ui_update
if capacity <= self.options['shutdown']:`
You have not mentioned that UPS (Pi Sugar or something) is connected. If none present and ups is switched on in config file (main.plugins.ups_lite.enabled = true) than pwnagotchi won't start.