evilsocket/pwnagotchi #1151: Initial Startup

velcro10, 2023-03-31:
So I am at a lose. I followed the guide step by step exactly and I cannot get any image to appear on my screen and the ras pi just blinks green.

My setup is a windows 10 machine, ras pi zero w, waveshare 2.13 e-paper hat-F2 screen V3 and an 8gb sandisk micro sd.

I followed the steps by flashing the SD with windisk though, BalenaEtcher seems to throw an oops error everytime it goes to validate the flash (yes, I am running it in admin mode). I then open file explorer and click on the SD card and create the "config.toml" and modify it for my screen version etc. 

I have flashed multiple times and still the same results. I even let the device stay connected to power for roughly 6 hours with still no results.

Any help would be great, thank you.

velcro10, 2023-03-31:
I am able to ping and change the IP of the device, just unable to see anything at all via the screen.

microbmen, 2023-04-14:
you need to update using the github repo which has V3 support; 
ui.display.type = "waveshare_3"

image 1.5.5 doesnt have V3 support.

closed 2024-03-01 by velcro10