evilsocket/pwnagotchi #1156: Please update release to 1.5.6(x) , need Waveshare_3 support

microbmen, 2023-04-14:
Waveshare V3 is all over Amazon, and many people are buying the V3 display.  Version 1.5.5 does not have support for V3 in it;
i.e. ui.display.type = "waveshare_3"

Please bundle a new image so that the display can be set to V3; The github repo does have support for _3, and everything works great when building fresh.  The image needs to be updated.  I think this will alleviate a lot of pain from many new comers to the utility;  

Thanks :-)

D337z, 2023-04-15:
Surprisingly, it actually DOES have it, but the image uses such old resources that it can't work properly.  Your best bet is to compile it from source, but even then it's going to take a few sleepless nights since the master source is...very broken to put it nicely.  It's based off of 2019 work, so it will take a massive update just to get waveshare_v3 working via an image file.

microbmen, 2023-04-15:
sadly, i know what you mean.  i built it from source. 

a much quicker method, which worked well was to take the 1.5.5 image, deploy it, then do a git clone, and `sudo pip3 install .` , which copied over the updated waveshare_3 files.    Worked perfectly.

HasseM, 2023-05-06:
This is indeed a major problem for new users. They buy a V2.1 waveshare assuming it is a V2 but then after lot's of struggling to get it to work they find out it is a V3 and no support at all....

Do we need to crowd-fund a V3 to the author for an updated image?

D337z, 2023-05-06:
I'm actually trying to build a fully updated version myself which includes the V3 support, updated and optimized binaries, and a working tensorflow for the ARMv6k+fp processor.  But I'm stuck at Bazel.  I'll try working with the main image and updating it as you did to see if it gives me anything more to work with, but the entire system needs updated and worked on to at least work on the intended hardware properly which is likely why it hasn't been done.

virtadpt, 2023-05-06:
It might be time for a fork, if not a straight-up scratch build with brand-new everything.

D337z, 2023-05-06:
That's partially what I'm working on.  I'm trying to make the 32-bit ARM compatible images, but I'm having to work from scratch on a few of the binaries since they only support ARCH64.  On the brighter side, I'm hoping to upgrade this to a 64-bit port to open up some speed for the AI training for supported systems.  Everything I'm finding warns me to only use pre-trained Tensorflow, but it defeats the entire purpose of the project unless reworked to capture, bring to multi-cored base (Pi3, Zero2, or 4), train, and upload the updated model to be used in the field once more.
On the bright side, I'm getting more optimal binaries out of it with newer versions of OpenJDK and GCC than the original afforded, but it's a slow process as I'm not great at cross-compiling for other architectures.