evilsocket/pwnagotchi #1161: [BUG]

hellomynameisgithubuser, 2023-05-10:
I have completely setup my pwnagotchi but when I plug it into my Windows 11 nothing happens. This is my first project with my raspberry pi zero w h so if yuo can please tell me what I should do.

MNarath1, 2023-05-10:
I assume you are trying to get ethernet over usb working? The driver required for that isn't part of windows driver repostory anymore so you can either try to download it from an archive site or use a linux vm with a hypervisor like virtualbox that allows you to passtrough the usb ethernet device that represents the pi to the vm so you can then use the linux guide to connect to the pwnagotchi trough getting internet to work is still a bit fiddly and the most consistent success i had for that is just using the bt plugin instead to allow it to use the internet connection of my phone

MNarath1, 2023-05-10:
You could also just setup bt-tether and call it a day and that works fine for the most part but if you try to then use the PAN on your phone to connect to the pwnagotchi using your computer its horribly unstable especially on ios so the vm option while slightly annoying and also a bit fidly is still the better option or getting the driver from a 3rd party for windows but i at least am more confortable with the vm solution since i don't have to trust a random 3rd party to not have tampared with the driver.