evilsocket/pwnagotchi #1162: Waveshare V3 and Pi4

Summ1tSunshine, 2023-05-12:
Could anyone help me out in getting pwnagotchi going on a Pi4 and a version 3 Waveshare? I have had a really hard time getting anywhere with this.

Thanks.

JaCKoP619, 2023-05-23:
https://pwnsec.io/how-to-make-a-pwnagotchi-in-2022-with-a-waveshare-v3-screen/
https://github.com/DrSchottky/pwnagotchi/releases
This beta version works, but it's not great.

Summ1tSunshine, 2023-06-06:
I actually have the exact hardware in the guide but it just seems to reboot over and over and I never get any data on the waveshare v3

JaCKoP619, 2023-06-16:
i tried once by setting it on config.toml file to waveshare v2 and then changing it via ssh in config file to v3. also languages setting isn't best idea on the first boot, it's best to do the basics and change it later