evilsocket/pwnagotchi #1166: [BUG]

anubis-84, 2023-05-29:
**Describe the bug**
Red screen, using Waveshare 2.13_v4 Three Color screen.

**To Reproduce**
Steps to reproduce the behavior:
1. Raspberry PI Zero W
2. Waveshare 2.13_v4 color screen
3. Last 1.5.5 Pwnagotchi Image

**Expected behavior**
After install the latest Pwnagotchi Image avaliable (1.5.5), the waveshare screen does not works. Appears blank, then black and after 10 seconds still red forever.

**Screenshots**
N/A

**Environment (please complete the following information):**
 - Pwnagotchi version 1.5.5
 - OS version (Pwnagochi)
 - Raspberry Pi Zero W
 - Waveshare 3-color screen 2.13 v4

**Additional context**
Tried a lot of stuffs...Including change the color to red and other waveshare's options.

densev1975, 2023-06-06:
waveshare oled raspberry hat not working , just white screen and occasional flickering

xNymia, 2023-07-16:
+1 same issue, no luck with any variations of Display and Colour.