evilsocket/pwnagotchi #1169: Pwnagotchi d0n't work after battery ran out of power

tXambe, 2023-06-20:
I have a pwnagotchi with pisugar battery. The battery ran out of power and now does not work, the screen is frozen, and when I connect it to a pc nothing happen, no detection for any pc (windows/Linux). I have reinstalled everything as indicated on the website of pwnagotchi but without result I don't know what else to try anymore.

I hope anyone can help me.

A greeting and thanks

Loader009, 2023-06-24:
Probably a dead rpi?
Try out some rpi default image with another sdcard and via usb cable - if it still doesn't work, it's most probably dead.

tXambe, 2023-06-29:
Hello,

Yes I try with other versions of pwnagotchi ( 1.5.3, 1.5.2), sd-card and don't work, how you say, probably is dead.

A greeting and thanks