evilsocket/pwnagotchi #1170: Is the project dead with the new Flipper zero?

Crash891, 2023-07-10:
hello, i hope this project goes on, but i don't see any update of the firmware (for example for the rpi 2w), and i'm worried that it is dead... i hope not

moni11811, 2023-07-10:
A Flipper Zero is $300?
I can do this project for $20?

This project is not dead - but it was considered 'complete' by the developer. 

No update or upgrade path was mentioned and it never will be as the dev "retired"... in the end, this is github everything is open source and we all collaborate to make something awesome. You can also copy-paste (within reason) this code and create your own Monster Zero if you decide...

Correct me if I'm wrong but isnt Flipper Zero a little closed source? Also does it even have built-in wifi?

As long as there are Arm64 machines and Github is still running this project lives on. 

As well as in our hearts.

dj1ch, 2023-07-13:
there is a 1.6.2 version in one of the forks of this which gives support for the 2w.

TasyDevilsky, 2023-09-25:
I recommend this fork :) https://github.com/aluminum-ice/pwnagotchi/releases Works very good for me !

debasser1911, 2023-09-25:
> I recommend this fork :) https://github.com/aluminum-ice/pwnagotchi/releases Works very good for me !

yeah, but it only for RPiZ2W