evilsocket/pwnagotchi #1175: New Installation. Fails to upgrade

kiraitachi, 2023-08-19:
**Describe the bug**
I have recently installed Pwnagotchi Image. Latest one as of 2023 1.5.5. After fixing the keyring issues for upgrading and connecting properly the device to internet it fails to upgrade because there are packages that break the kernel.
**To Reproduce**
Steps to reproduce the behavior:
1. `sudo apt update`
2. `sudo apt upgrade`
3. ```
The following packages have unmet dependencies:
 kalipi-kernel : Breaks: raspberrypi-kernel but 1:1.20230509~buster-1 is to be installed
```

**Expected behavior**
To upgrade

**Screenshots**
![image](https://github.com/evilsocket/pwnagotchi/assets/40744673/aa82866b-8097-4d67-b79a-dbab4d2801ed)

**Environment (please complete the following information):**
 - Pwnagotchi version 1.5.5
 - Type of hardware: Pi-zero WH

Anyone got it fixed?

MegaTheLEGEND, 2023-08-21:
I don't have an answer to your issue. But I will say that [wpa-2's fork](https://github.com/wpa-2/pwnagotchi/releases) of Pwnagotchi worked for me on my pi-0w. It includes support for waveshare v3 screens and fixed the AI mode not starting. [These instructions](https://github.com/Xyl0se/Pwnagotchi-new-guerilla-guide) are also very helpful.

kontrakote, 2023-08-22:
Try this:

`wget https://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/raspberrypi-bootloader_1.20230405-1_armhf.deb`

`wget https://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/raspberrypi-kernel_1.20230405-1_armhf.deb`

`wget https://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/raspberrypi-kernel-headers_1.20230405-1_armhf.deb`

`dpkg -i ./raspberrypi-bootloader_1.20230405-1_armhf.deb`

`dpkg -i ./raspberrypi-kernel_1.20230405-1_armhf.deb`

`dpkg -i ./raspberrypi-kernel-headers_1.20230405-1_armhf.deb`

Got it [here](https://wiki.kewl.org/dokuwiki/tools:raspbian).

wpa-2, 2023-08-24:
> Try this:
>
> `wget https://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/raspberrypi-bootloader_1.20230405-1_armhf.deb`
>
> `wget [...]

That will break it and is pointless.