evilsocket/pwnagotchi #1178: Question about /handshakes#

azrael-tzel, 2023-09-28:
First of all, i really love this pwnagotchi project, and i find it very amusing and educative. 

Now i run the ssh in terminal. 

I managed to get the root 

Found the acces to the handshakes, at least it seems so. 

It went from 
Root@pwnagotchi:/home/pi#
To
Root@pwnagotchi :~/ handshakes# 

Now i cant find what command to type next. 

Maybe i just look over something here and miss it out. 

But if someone could help me a bit that would be nice. 

I love to learn

Nannk, 2023-09-28:
Hi
what do you want to do?

azrael-tzel, 2023-09-28:
Hello, i want to run the command from this setting to reveal and save the handshakes trough ssh in my terminal

azrael-tzel, 2023-09-28:
What commands do i run if i want to use  what youre saying? 

Can you give an example?

Thanks alot

Nannk, 2023-09-29:
To list directory contents you can use `ls` check `ls --help ` for available flags. 
For openssh copy you can use `scp [OPTION] [user@]SRC_HOST:]file1 [user@]DEST_HOST:]file2` ([source](https://linuxize.com/post/how-to-use-scp-command-to-securely-transfer-files)), though im not sure if it would work with a windows machine.
I would advise to check pwnagotchi tutorials, for example [this one](https://www.youtube.com/watch?v=P74qRNETQXs), it explains how to use scp with windows.

azrael-tzel, 2023-09-29:
Thankyou, will try.