evilsocket/pwnagotchi #1182: ssh permission denied (publickey)

TheChristographer, 2023-10-16:
hi, im trying to set my device up for the first time and when i fist ssh'ed into 10.0.0.2 it started to go right through i typed yes and then it said ssh permission denied (publickey). now everytime i try to to ssh in it just says that over and over again. any help would be appreciated and im new to this sort of stuff so take it easy on me. Thank you

TheChristographer, 2023-10-16:
UPDATE: i resolved the above issue and ssh'ed into the pwnagotchi but when i jump over to filezilla to finish the config it just says access denied and Connot connect

timmosaurusrex, 2023-10-16:
If you're not already using the  root user, try logging in as root. First you need to enable root ssh login in the /etc/ssh/sshd_config. Change PermitRootLogin from no to yes. If you did everything correctly you can now login as root.

anasbousselham, 2023-11-05:
Hi, I have the same issue with the credential : pi@10.0.0.2 and pwd: raspberry is wrong. (permission denied )
Please can you provide me the right way to connect to my pizero w2.
Thnaks

sizzlik, 2023-11-06:
> Hi, I have the same issue with the credential : pi@10.0.0.2 and pwd: raspberry is wrong. (permission denied ) Please can you provide me the right way to connect to my pizero w2. Thnaks

Got the same problem..fresh install, first boot shows new ethernet, changed to static ip, pinged to test and got answer, tried to ssh in, accepted the host request with yes, tried passwrod: raspberry, permission denied, try again. Let me know when you figure it out.

TheChristographer, 2023-11-06:
I got in to the terminal no problem, my issue was with FileZilla. Have you
watched the talking squatch video on it?

sizzlik, 2023-11-06:
i replyed to anasbousselham above

timmosaurusrex, 2023-11-06:
> @timmosaurusrex 
> I did that part before I tried using FileZilla to finish the config but that's when I get an access denied trying to login as-
> 10.0.0.2
> Username-Root
> Password-password
> Port-22
> I can login through the terminal as pi and switch to root but it won't let me connect in FileZilla. Any insight into why?
> Also could I just finish setup by editing the "config.toml" directly on the sd and then plug it back in?

I hope i understand the issue correctly. Try connecting using ssh en switch to root user. After that edit the /etc/ssh/sshd_config using your favorite text editor. I use nano, so for me it will be "nano /etc/ssh/sshd_config". Go to the line where you see something with "permit root login" and edit it to yes. Quit nano and restart your pi or restart the sshd service. You should now be able to login as root in FileZilla.

briss87, 2024-04-30:
I encountered the same problem as you, once I entered the "sshd_config" file I cannot find the "PermitRootLogin" entry and this does not allow me to log in with FileZilla.