evilsocket/pwnagotchi #1183: First start takes too long and never ends

notnotbx0, 2023-11-01:
I am having issues on booting my pwnagotchi while following the website steps. After writing the image on the SD Card and adding the `config.toml` file, it never starts.

I have left the device conected to my computer for over 8 hours and it didn't show anything on screen nor connected to my computer. What can be done in this situation? I'm running it on a Raspberry PI Zero W, with Waveshare 2.13 (white, black, red).

My config.toml file is like this:
```
main.name = "pwnagotchi"
main.lang = "en"
main.whitelist = [
  "mynetwork",
]

main.plugins.grid.enabled = true
main.plugins.grid.report = true
main.plugins.grid.exclude = [
  "mynetwork",
]

ui.display.enabled = true
ui.display.type = "waveshare213d"
ui.display.color = "black"
```

Is this an recurring issue or am I missing something?

notnotbx0, 2023-11-02:
Fixed by using the [1.5.5fix](https://github.com/wpa-2/pwnagotchi/releases)

derrickr, 2023-12-05:
Been having the same issue. Tried new hardware = same problem.
Now trying the [1.5.5fix](https://github.com/wpa-2/pwnagotchi/releases) but it still seems to be taking a long time.
It's been about 15 minutes so far. Do you know how long a first boot should take?

psiyco, 2024-01-09:
i also have this if i dont ssh and confirm fingerprint it will never load ... you cheacked your ssh fingerprint ? try to delete tham and confirm the new ones

TheGroundZero, 2024-04-11:
I'm having the same issue.
Flashed SD card multiple times to far. Even left the RPi powered via the power-only USB-port overnight to it definitely got enough time to setup and generate SSH keys.
Connecting to my Windows systems only shows a COM port, no network port.