evilsocket/pwnagotchi #1186: Can ping 10.0.0.2, but get ssh refused

jbrepogmailcom, 2023-11-12:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Installation as in guide
2. ping pi@10.0.0.2, ping responds
3. ssh pi@10.0.0.2
ssh: connect to host 10.0.0.2 port 22: Connection refused

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
 - Pwnagotchi version: latest
 - OS version
 - Type of hardware Raspberry Pi Zero W
 - Any additional hardware used

**Additional context**
Add any other context about the problem here.

nickba86, 2023-11-23:
Hi, in order to solve this I recommend to follow this guide:
https://github.com/Xyl0se/Pwnagotchi-new-guerilla-guide

2. Basic Connectivity (SSH, FTP, Connection Sharing)
2.1 Connect to PC
Because I am running Windows 11 as a daily driver, this section will cover only Windows. For establishing a connection on Linux or MacOS refer to the [official guide](https://pwnagotchi.ai/configuration/).

connect RP0W data port to pc (Micro-USB to USB A)
wait for the device to boot up for the first time (20+ Minutes)
Check Device Manager for COM-Port - [as described here](https://learn.adafruit.com/turning-your-raspberry-pi-zero-into-a-usb-gadget/ethernet-gadget#if-you-are-using-windows-as-the-host-machine-2570572)
If Windows didn't install the RP0W as an "Ethernet Gadget", download the corresponding driver on [Windows Update](https://www.catalog.update.microsoft.com/Search.aspx?q=USB+RNDIS%20Gadget)
Unpack the CAB
In Device Manager select the COM-Device and update the driver with the one from the CAB
make sure to check in Network Devices for the Interface and configure TCP/IP v4 to use 10.0.0.01 as its IP-address, 255.255.255.0 as Subnet and 10.0.0.1 as Gateway.

It worked for me in the first atemp
Regards

virtadpt, 2023-11-23:
Do an `ssh -vv pi@10.0.0.2` and post what you see into a comment.

Does your box have a route for that addres?  `route -n`

What is your addressing like at home?  Does it overlap with 10.0.0.2?

How, exactly, are you trying to connect to it?  Do you have a USB Ethernet adapter or something?

haotianh9, 2024-05-14:
I

> Do an `ssh -vv pi@10.0.0.2` and post what you see into a comment.
> 
> Does your box have a route for that addres? `route -n`
> 
> What is your addressing like at home? Does it overlap with 10.0.0.2?
> 
> How, exactly, are you trying to connect to it? Do you have a USB Ethernet adapter or something?

I have a similar problem, and here is my reply:

OpenSSH_8.6p1, LibreSSL 3.3.6
debug1: Reading configuration data /Users/user/.ssh/config
debug1: Reading configuration data /etc/ssh/ssh_config
debug1: /etc/ssh/ssh_config line 21: include /etc/ssh/ssh_config.d/* matched no files
debug1: /etc/ssh/ssh_config line 54: Applying options for *
debug2: resolve_canonicalize: hostname 10.0.0.1 is address
debug1: Authenticator provider $SSH_SK_PROVIDER did not resolve; disabling
debug1: Connecting to 10.0.0.1 [10.0.0.1] port 22.
debug1: connect to address 10.0.0.1 port 22: Connection refused