evilsocket/pwnagotchi #1187: No display and No start with Waveshare V4

nickba86, 2023-11-17:
Hello Everyone!!
I hope you are doing well

**Describe the bug**
The screen does not show Anything.
The: "systemctl status pwnagotchi" Shows:
[image]

The Log Shows:
[image]

**To Reproduce**
Steps to reproduce the behavior:
1.  Make a Fresh copy of the pwnagotchi ISO in a SD card.
2. ... Connect to the pwnagotchi with FTP and replace the files with the last version on github
[image]
That's because the image available to download does have the latest changes with the support for the last type of screen 
4. Modify the CONFIG.TOML with:
ui.display.type = "waveshare213inb_v4" 
I understand that's the way to it because in the init we have the last modification:
[image]

**Expected behavior**
I want to see the screen working with image.

**Environment (please complete the following information):**
 - Pwnagotchi version: Last image available: [v1.5.5](https://github.com/evilsocket/pwnagotchi/releases/tag/v1.5.5) 
 - Type of hardware: [Raspberry Pi Zero W](https://www.raspberrypi.org/products/raspberry-pi-zero-w/) and Waveshare 2.13 Inch e-Paper Display HAT V4 250x122 Resolution E-Ink Screen Electronic Paper Module with Embedded Controller Partial Refresh for Raspberry Pi/Jetson Nano/Arduino,Support SPI Interface

Thank you very much!!!!!

Necr0byte, 2023-11-17:
Out of curiosity, can you try running `journalctl -u pwnagotchi.service` when you start it to see if it's writing anything to stderr before crashing?

spitecho, 2023-11-17:
I had the same issue. I think it's because the most recent image (version 1.5.5, released April 18, 2021) doesn't have the Waveshare v4 update (released January 18, 2023).

I fixed it by copying the git cloned pwnagotchi folder and binary file from this repo:

```
ssh pi@10.0.0.2
sudo systemctl stop pwnagotchi.service
git clone https://github.com/evilsocket/pwnagotchi.git
cd pwnagotchi
sudo mv /etc/pwnagotchi /etc/pwnagotchi.bak
sudo cp -r pwnagotchi /etc/pwnagotchi
sudo mv /usr/local/bin/pwnagotchi /usr/local/bin/pwnagotchi.bak
sudo cp bin/pwnagotchi /usr/local/bin/pwnagotchi
```

Then I edited the /boot/config.toml to have 'ui.display.type = "waveshare213inb_v4"' and rebooted. This brought the screen up, but with flashing. Then I edited the /etc/pwnagotchi/config.toml to:
```
ui.display.type = "waveshare_3"
ui.fps = 0.5
```

Setting it to "waveshare_3" (despite the screen being labeled v4) removes the constant screen flashing and everything works as expected.

nickba86, 2023-11-17:
Hi all !!! Thanks for the answer!! I am still facing issues:
@Necr0byte , the log does not says much:
[image]

@spitecho , I am really sorry but I am afraid I was able to follow your instructions. Please help me to understand what I'm doing wrong:
1. "Then I edited the /boot/config.toml to have 'ui.display.type = "waveshare213inb_v4""
I was not able to find the config.toml in the boot
[image]
As you can see in the image there no config.toml in boot.

2./etc/pwnagotchi/config.toml  
In this case I was able to modify but I am confuse because I understood there I should put 'ui.display.type = "waveshare213inb_v4"'

3.  In this path I put the last version:/usr/local/src/pwnagotchi
[image]

4. I did not know why doing this:
sudo mv /etc/pwnagotchi /etc/pwnagotchi.bak
sudo cp -r pwnagotchi /etc/pwnagotchi

It like putting the files in folder with the config???

Thank you very much for the help!!!!!!!!!

spitecho, 2023-11-18:
You create the /boot/config.toml yourself and add the configuration lines to it. The lines are read at next boot, saved to /etc/pwnagotchi/config.toml, after which, the /boot/config.toml file is removed.

As for the rest, you're basically replacing one file (/usr/local/bin/pwnagotchi) and one folder (/etc/pwnagotchi/) with the updated versions from this project's main page.

LookaD, 2023-11-18:
hi @spitecho im also having the same issue, i have a question about your method,
i installed with the image found in the release, so i dont have all the files inside of  the pwnagotchi that are in the updated repo,
so im not sure its working correctly.

i mean the image doesnt have all the .py files etc..

spitecho, 2023-11-18:
@LookaD @nickba86 My mistake, I misremembered the location to copy the repo files to. Instead of "/etc/" the pwnagotchi folder from this repo should be copied to "/usr/local/lib/python3.7/dist-packages/".

LookaD, 2023-11-18:
@spitecho ok so i did as you said, but i suppose im still missing something doing journalctl -u pwnagotchi.service is giving these errors

so i mean,
updating the pwnagotchi folder in  "/usr/local/lib/python3.7/dist-packages/"
and the file in /usr/local/bin/

i also did chmod 755 to make it as same as the original and chown for root:root

[image]

Thank you for helping btw!

nickba86, 2023-11-20:
Thank you very much @spitecho.
In order to make it work I had to follow your instructions. I will recap for others:
1. Update the  pwnagotchi after clone the image 1.5.5
2.  Follow the Guerilla Guide: https://github.com/Xyl0se/Pwnagotchi-new-guerilla-guide
3. Update the code: /usr/local/lib/python3.7/dist-packages/" (With the code in Github the last image does not have the last version).
[image]
[image]
4. Make a config.toml and put in the boot
[image]
In this case you must put
[image]
5. In /etc/pwnagotchi  you put the other config.toml
[image]
in this case you must put: ui.display.type = "waveshare_3" in the config.toml
6. Enjoy

THANKS!!!!!!!! :) 👍

closed 2023-11-20 by nickba86