evilsocket/pwnagotchi #1200: [BUG]

darkt1981, 2024-01-03:
I have everything running and i connect to the device in man and auto/ai mode.
But: i cannot get the screen to run:

I have this set:
https://www.waveshare.com/product/2.13inch-touch-e-paper-hat-with-case.htm

i set it up as a V4 seeing the sticker says its V4.
"waveshare213inb_v4"

i also tried as waveshare_3 still nothing, i check if the "v4 patch was in the files and yes it is.

also can some explain something to me the ui.fps when i set this to like 1 how long is that in Reallife because i dont see that documented anywhere?

MagicalMutation, 2024-01-03:
so..

I checked everything over and over.. then gave up. had a few whsikys..  then ordered another screen..

it was the screen... dodgy ribbon cable..

works great.. 

just got to figgure outn ow how to get it to display cracked keys and im on to a winner...

darkt1981, 2024-01-04:
> so..
> 
> I checked everything over and over.. then gave up. had a few whsikys.. then ordered another screen..
> 
> it was the screen... dodgy ribbon cable..
> 
> works great..
> 
> just got to figgure outn ow how to get it to display cracked keys and im on to a winner...

there is a plugin for that 
https://github.com/evilsocket/pwnagotchi-plugins-contrib/blob/master/display-password.py

megatnuar, 2024-01-10:
i followed this reddit post and it works go have a try https://www.reddit.com/r/pwnagotchi/s/Ud2580MWsJ

darkt1981, 2024-01-17:
Issue solved:

https://www.waveshare.com/product/2.13inch-touch-e-paper-hat-with-case.htm V4

use "waveshare_3" as displaydriver

Seems i got a double wammy, first they send me the wrong 1 instead of 2 w, then it seems the screen e-paper was broken replaced it ans started on the first run.