evilsocket/pwnagotchi #1205: Wavesahre e-papre 2.13inch does not work

cohen-sydney, 2024-01-09:
**Describe the bug**

2.13 e-per wavshare v2 does not work
on RpI ero W1 - with paper HAT
**To Reproduce**
Steps to reproduce the behavior:
1. plugged in on top
2. have the config file set
ui.display.enabled = true
ui.display.type = "waveshare_2"
ui.display.color = "black"

no display

is working as I can see it in the internal URL @socket 8080
**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
 - Pwnagotchi version 1.5.5
 - OS version 
 - Type of hardware RPI Zero w V1
 - Any additional hardware used
1.13" paper hat from wavershare v2
**Additional context**
Add any other context about the problem here.

cohen-sydney, 2024-01-09:
I used this build and V4 for the waveshare device all good now.

would be nice to have a new build here. thanks

JuergenEgger, 2024-01-21:
ui.display.enabled = true
ui.display.type = "waveshare 2" ? what exactly does your entry look like, I also have the problem V4 display and nothing works...
ui.display.color = "black"

widhalmt, 2024-01-31:
I got the same problem. Also with a Raspberry Pi Zero WH. Just checked the configuration files. Entry in `config.toml` is identical to what's in `default.toml`.

widhalmt, 2024-01-31:
I have a v4 of the waveshare 2.13" and the changes in #1135 worked perfectly for me.