evilsocket/pwnagotchi #1206: Bluetooth tethering opening bettercap and not pwnagotchi webbui

psiyco, 2024-01-09:
**Describe the bug**
pwngotchi working and configured can acsses all on pc 
after connecting bluetooth im going to 192.168.40.40 on my phone and it bringing me to bettercap web ui 
evreything else is not working no matter how i play with my port at the end its or bettercap web ui or "this site cant be reached"

pwnagothci 1.5.6beta2 ( https://github.com/DrSchottky/pwnagotchi/releases/ )

*note bettercap is first to open on pc if i type 10.0.0.2 without 8080 port 
**To Reproduce**
Steps to reproduce the behavior:
1. run pwnagotchi 
2. bluetooth auto connect (showing status -c)
3. type 192.168.40.40 on my phone opening bettercap ui

**Expected behavior**
open pwnagotchi webui 

**Screenshots**
can provide logs if asked and instructed where to get from 

**Environment (please complete the following information):**
 - Pwnagotchi version - 1.5.6 beta2 
 - OS version - windows 11  (yes i found a stable soulution to share connction and update the pi with no packet loss)
 - Type of hardware galexy s21 ultra 
 - Any additional hardware used

**Additional context**
would be great if i could tell pwnagotchi of web ui loading priority like with bluetooth scan 
than it will load the wanted webui first  (but in not a programer so dont listen to me ) 

thanks in advance for bringing us such an amazing project <3

psiyco, 2024-01-09:
i will try to reproduce it with 1.5.5 
will see if it behaves differently

T-Proc, 2024-01-31:
The UI is on port 8080 not 80, bettercap runs it's api on 80
try 192.168.40.40:8080