evilsocket/pwnagotchi #1209: Ai enabling

OutsiderSV, 2024-01-12:
I'm pretty new to this so bare with me, I recently built my pwnagotchi and I have been reflashing trying different configurations, I am currently on version 1.5.5 but have tried beta 1.5.6 v2 the 1.5.6 does not start capturing anything while the 1.5.5 that I have currently does not have working ai, I checked the logs and this is the error I receive for the AI
[2024-01-12 04:19:11,219] [ERROR] error while starting AI (numpy.ndarray size changed, may indicate binary incompatibility. Expected 44 from C header, got 40 from PyObject)
Ive tried sudo pip3 install --upgrade numpy like other forums suggested but my cmd prompt keeps getting caught up at this point
Running setup.py bdist_wheel for numpy ... /

I have left it at that for 2 hrs and still nothing
any help is appreciated

kontrakote, 2024-01-12:
What RPi do you use?

OutsiderSV, 2024-01-12:
> What RPi do you use?

Sorry I have a RPi 0 WH
with a waveshare v4 
and pisugar 3

kontrakote, 2024-01-12:
So strange. 1.5.6 beta v2 should work on RPi 0 WH.

OutsiderSV, 2024-01-12:
> So strange. 1.5.6 beta v2 should work on RPi 0 WH.

when I do the beta v2 it doesnt scan networks at all and the bottom left displays 0(0) and never updates

renewedbullet, 2024-02-02:
> > What RPi do you use?
> 
> Sorry I have a RPi 0 WH with a waveshare v4 and pisugar 3

should look at some of the newer pwnagotchi repos 2.6.4 torch in mind

gagregrog, 2024-02-16:
@OutsiderSV likely it is still building when you leave it there even though it doesn't look like anything is happening. The pi zero is resource constrained, so building the wheel will take ages.

My advice: increase swapsize and build in verbose mode.

Here's some notes I wrote when I got v1.5.5 working on my pi0:

### Increase Swapfile

Downloading `numpy` will require it to compile on the RPi which will take a very long time. In order not to go insane we need to bump the swapfile so that the Pi has more memory. The process outlined below comes from [here](https://nebl.io/neblio-university/enabling-increasing-raspberry-pi-swap/).

1. Stop the swap: `sudo dphys-swapfile swapoff`
2. Change the `CONF_SWAPSIZE` from `CONF_SWAPSIZE=100` to `CONF_SWAPSIZE=1024` in `sudo nano /etc/dphys-swapfile`
3. Initialize the swapfile with `sudo dphys-swapfile setup`
4. Start the swap with `sudo dphys-swapfile swapon`

After building `numpy` go back and set the swap back to 100.

### Upgrade numpy

Now that we have increased the swapsize we can finally upgrade numpy. This will take a very long time, so make sure to install it using `--verbose` so that we can monitor the progress.

1. SSH into the pi
2. Start a screen session so that if we get disconnected for some reason we can get back into it: `screen bash` (to exit, ctrl-a and then d. To rejoin `screen -r`. To kill the session ctrl-d)
3. Start the install: `sudo pip3 install --verbose --upgrade numpy --no-cache-dir`
4. Once it successfully installs, go back and reset the swap size

You should be good to go now!
Restart the service and boot it into Auto mode and watch the logs and you should see that the AI model now loads after about 20 minutes: `sudo touch /root/.pwnagotchi-auto && systemctl restart pwnagotchi && tail -f /var/log/pwnagotchi.log`

hashansr, 2025-02-17:
The above instructions unfortunately didn't work for me :( I am on a rPI Zero 2W using a fast SD card (Sandisk Extreme Pro 128GB). My pwnagotchi still won't switch from Auto to AI