evilsocket/pwnagotchi #1213: [BUG] error when i follow "install on any gnu/linux

naxil, 2024-02-08:
**Describe the bug**
Step3 of the guide command sudo pip3 install -r requirements.txt

**To Reproduce**
Steps to reproduce the behavior:
1. Lubuntu 22.04 kernel 6.2.0-26
2. Follow the site guide
3. Step 3 install pwngotchi (all version 1.5.5 and latest master)

**Expected behavior**
Probably finish the installation without error

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
 - Pwnagotchi version latest
 - OS LUBUNTU 22.04 x64
 - Type of hardware
 - intel atom hdmi pen called stck1a32wfc

**Additional context**
Iam noob, probably iam not understand some part, but i follow step by step the guide... 
Thank you for helpme

Paste error on = https://www.mediafire.com/file/qtyeixiyi0z9r9q/log.txt/file/ 
Huge 5 mb of log... maked with --log on pip3

MarcDoughty, 2024-02-29:
I believe some of the packages that are in that 'requirements.txt' file are old enough where they no longer work with newer Python versions on most 2022+ distros.

I ran into the same thing and ended up down quite a rabbit hole of trying to unstick dependencies, to no avail (but I am also no expert).

@evilsocket - let me know if you want me to share a few requirements.txt reworks that got at least as far as allowing the installation and compiling of pwnagotchi under Python 3.11+. I still haven't got it p and running yet, but I think I'm closer than @naxil .

naxil, 2024-03-01:
If u can share with me your requirements.txt iam really happy! Do you finish the compiling?

mrfoxie, 2024-06-24:
> I believe some of the packages that are in that 'requirements.txt' file are old enough where they no longer work with newer Python versions on most 2022+ distros.
>
> I ran into the same thing and [...]

yes can you share new `requirements.txt` it will be really helpful for us