evilsocket/pwnagotchi #1214: Pwnagotchi web UI not working

Soraya-sky, 2024-02-11:
Hello everyone ! 
I've been struggling a bit with my pwnagotchi. After 4 days on it, I'm still no able to connect to the web UI.

I run the pwnagotchi on a raspi0w, without display.
My config.tolm seems to be working as the name of my pwnagotchi have been successfully changed. Although it's a bit weird because it's not always removed from the /boot directory on the SD card.

Here is my config.tolm file :
-----------------------------------

main.name = "mychosenname"
main.lang = "en"
main.whitelist = [
  "4c:34:88:1a:9f:c8",
  "fo:od:ba:be:fo:od",
  "fo:od:ba"
]

main.plugins.grid.enabled = true
main.plugins.grid.report = true
main.plugins.grid.exclude = [
  ""
]

 ui.display.enabled = false
 ui.display.type = "waveshare_2"
 ui.display.color = "black"

main.plugins.bt-tether.enabled = true

main.plugins.bt-tether.devices.android-phone.enabled = true          # the name of this entry is android-phone
main.plugins.bt-tether.devices.android-phone.search_order = 1         # in which order the devices should
                                                                      ## be searched. E.g. this is #1
main.plugins.bt-tether.devices.android-phone.mac = "XX:XX:XX:XX:XX:XX" # you need to put your phones
                                                                      ## bt-mac here (settings > status)
main.plugins.bt-tether.devices.android-phone.ip = "192.168.44.44"     # this is the static ip of your pwnagotchi
                                                                      ## adjust this to your phones pan-network
                                                                      ## (run "ifconfig bt-pan" on your phone)
                                                                      ## if you feel lucky,
                                                                      ## try: 192.168.44.44 (Android) or
                                                                      ## 172.20.10.6 (iOS)
                                                                      ## 44 is just an example, you can choose
                                                                      ## between 2-254 (if netmask is 24)
main.plugins.bt-tether.devices.android-phone.netmask = 24             # netmask of the PAN
main.plugins.bt-tether.devices.android-phone.interval = 1             # in minutes, how often should
                                                                      ## the device be searched
main.plugins.bt-tether.devices.android-phone.scantime = 10            # in seconds, how long should be searched
                                                                      ## on each interval
main.plugins.bt-tether.devices.android-phone.max_tries = 10           # how many times it should try to find the
                                                                      ## phone (0 = endless)
main.plugins.bt-tether.devices.android-phone.share_internet = false   # set to true if you want to have
                                                                      ## internet via bluetooth
main.plugins.bt-tether.devices.android-phone.priority = 1             # the device with the highest
                                                                      ## priority wins (1 = highest)

main.plugins.bt-tether.devices.ios-phone.enabled = false              # next device...
main.plugins.bt-tether.devices.ios-phone.search_order = 2

ui.web.enabled_ = true
ui.web.username = "username"
ui.web.password = "password"
ui.web.address = "0.0.0.0"
ui.web.origin = ""
ui.web.port = "8080"
ui.web.on_frame = ""

-----------------------------------------

- I have flashed the 1.5.5 version on the sd card, It has been recognised by the computer as a Ethernet/RNDIS gadget. Then I set the ip etc. on 10.0.0.1, 255.255.255.0 and 8.8.8.8 (see picture).
[image]

The only strange thing about that is that my computer is creating a new network almost each time I plug the pwnagotchi (network4,network5, etc.)

- I'm able to ssh with Putty, the pwnagotchi.local redirects to bettercap as it shoud and everything seems alright on that side. But the web UI stays stuck on "the site can't be reached" message. I tried multiple ip and ports, but nothing works.
- I also tried connected the pwnagotchi with bluetooth to my android phone, the connexion worked but the webUI would not work neither this way.

- The pwnagotchi seems to be working normally when not plugged to the computer as it has already some handshakes.

Does anyone have an idea of the problem ?
Thanks for the help

mieg, 2024-02-12:
In the line
`ui.web.enabled_ = true`
i would delete the _ at the end of enabled

sosdos, 2024-02-16:
are you able to connect to the bettercap ui at 10.0.0.2? 

i had the issue that while I could connect to bettercap, I couldn't connect to the actual ui using pwnagotchi.local:8080.

solved by just using 10.0.0.2:8080 to get there instead

sosdos, 2024-02-16:
maybe try removing everything about the web ui from your config file (except ui.web.enable = true)

if that doesn't work, try going back to the stock config file without any edits.

Soraya-sky, 2024-03-02:
> maybe try removing everything about the web ui from your config file.
> 
> if that doesn't work, try replacing the text in your config file with this (default config file):
> 
> main.name = "pwnagotchi" main.lang = "en"
> 
> main.plugins.grid.enabled = true main.plugins.grid.report = true
> 
> ui.display.enabled = true ui.display.type = "waveshare_2" ui.display.color = "black"

Thanks, m’y computer is broken right now, but I’ll try this asap.

wichi92, 2024-06-10:
Hi @Soraya-sky any updates on this? I have the same issue like you.

xiaoyao9184, 2024-06-22:
try 

check process
```
ps -aux | grep pwnagotchi
```

check service `Active: active (running)`
```
systemctl status pwngrid-peer.service  --no-pager --quiet
systemctl status bettercap  --no-pager --quiet
systemctl status pwnagotchi  --no-pager --quiet
```

check config no error
```
pwnagotchi --print-config
```

try run manual
```
/usr/bin/python3 /usr/local/bin/pwnagotchi --manual
```

Soraya-sky, 2024-08-02:
Hi !
Thanks for the help.

So I was away from this project for a while but I juste came back to it.
First thing I did was download the latest version, then reflash it on the same sd card after formating it.
I plugged the pwnagotchi on power port and let it initialize for a while.
I then plugged it on data port and started ssh.

And from there, I used :
sudo touch /root/.pwnagotchi-auto && systemctl restart pwnagotchi

To restart it in auto mod.
And I've been able to access the webUI.

Now something that is still weird is that the config.tolm file seems to not be read proprely as the name hasn't been changed...

I'll keep investigating this and I'll let you know when I find something.

VEYSkub, 2024-12-17:
the same problem. however, when trying to launch manually, there is a python error. what's on the official file, what's on the user (v 2).