evilsocket/pwnagotchi #1218: Pwnagotchi doesn't work

creatorpanda, 2024-02-23:
Hello :)

I bought a Raspberry Pi Zero 2 W that normally boots in RPi OS. peripherals, power LEDs, and HDMI work perfectly.

When I flashed pwnagotchi-raspbian-lite-v1.5.5 on my SD card and plugged it in, the device didn't do anything. The LEDs don't light up and the Waveshare screen doesn't work.

With the pwnagotchi-raspbian-lite-v1.5.5 SD on, the device doesn't work even with the screen unplugged, even before I soldered the pin header, so it's probably not a hardware issue.

Any suggestions? I want to make this project work!

Thank you very much :)

manne2400, 2024-02-24:
Try follow this guide : https://pwnagotchi.org/getting-started/index.html
I'm running a Rasp pi zero 2 w like you, and i had no trouble installing this image: https://github.com/jayofelony/pwnagotchi-bookworm/releases/tag/v2.8.3

creatorpanda, 2024-02-24:
Hello, manne2400! :)
thank you for responding!

> I'm running a Rasp pi zero 2 w like you, and i had no trouble installing this image

The image you suggested worked and the light now turns on.
for the first ten minutes, it was blinking, probably doing its thing. 
Now it has completely stopped blinking and is steadily on.

I am trying to make it work with a screen on it.
Before I first booted the RPi, I changed `ui.display.enabled = false` to **true**. 
It's still not working. 

I am completely n00b to such projects and I have an embarrassing question:
I need to clarify that I am not using a PiSugar and have just connected the screen on the soldered pin header. I think I've seen Pwnagotchis run fine without one, but maybe the pics were deceptive. Do I **need** one to power the screen?

Thank you very much!

manne2400, 2024-02-24:
Hello creatorpanda
Im a noob to, but i wil try and help you as good as i can.
What screen do you have ? Maybe this will help you: https://pwnagotchi.ai/configuration/#select-your-display
Did you choose the right one in config.toml:
ui.display.enabled = true
ui.display.type = "waveshare_2"  #Choose the display you have.
ui.display.color = "black"

Well i dont have screen on my raspberry pi zero 2 w, but i do on my raspberry zero w, it is working and get the power from the raspberry, as you can see in the config i have a wavashare 2.

I hope this helps you, btw on my raspberry pi zero 2, i'm using my phone as screen with a bluetooth connection.

creatorpanda, 2024-02-26:
Hello again :)

I am sorry for the delayed answer.
I have been trying what you told me, yet nothing worked. I will keep working on the info you sent me and if it all works I will close the ticket. The bluetooth scren on my phone sounds super interesting too, given the fact that concealment-wise, the device can hide at the bottom of a backpack, while someone looks innocently at their phone screen.

To be honest, the screen showed black spots only once I disconnected my RpiZ2 from my PC, but nothing else before that.
I will keep trying to make it work and hopefully, someone who has figured it out the waveshare screen will also comment here.

Thank you very much again! :)

bryzz42o, 2024-02-28:
Don't worry we're all different levels of n00b, I'd just try to get it set up and connected to your phone before config screen BC it is a much better UI and just easier and less sus in public Try and list all the hardware in as much detail as you can about the screen too as waveshare has alottt of variants/sizes :) my advice would be to
1. Remove the screen & reflash the SD (as annoying as it is) with a freshly downloaded .IMG from Jayfelonys repo linked above with BalenEtcher not RaspberryImager
2. Connect it to your computer, power up and leave it be for 5-10 mins to make sure it sets itself up properly
3. Connect it via HDMI and after boot seq the User is Pi & Pass is Raspberry, ethernet would prob be better as will allow it to connect to the internet if needs to pull anything during boot
4. Once your in run Sudo apt update & upgrade then sudo raspi-config and change the boot settings so the pwnagotchi boots straight in and ready with the Console Auto-Login selection.
If you get that all done you can pair to your phone with the bluetoothctl cmd's but make sure to add your Bluetooth MAC address to the android or iOS section of config.toml