evilsocket/pwnagotchi #1223: Lack of info.. not you guys.. lol

ALoGeNo, 2024-03-23:
Okey, in fisrt place thankyou for this great so great greatfull thing and your time of course..., and then it maybe be open a precedent or not but will be a point to know wich python version was used to build the proyect, what do you think?.

thank you guys !! a lot! sorry for my bad english! cheers!

wpa-2, 2024-04-04:
> o

There is a some good information here 
https://pwnagotchi.org/

franciscojs12, 2024-04-13:
There are two Pwnagotchi websites: https://pwnagotchi.ai/ & https://pwnagotchi.org/ which one is the official one? I see only the .ai one listed in this repo...

wpa-2, 2024-04-13:
.ai is official but .org is most updated and recent. 

This repo hasn't been updated in a while so I would avoid this image.

wpa-2, 2024-04-13:
> There are two Pwnagotchi websites: https://pwnagotchi.ai/ & https://pwnagotchi.org/ which one is the official one? I see only the .ai one listed in this repo...

.ai is official but .org is most updated and recent.

This repo hasn't been updated in a while so I would avoid this image.