evilsocket/pwnagotchi #1228: Win10/Win11 missing RNDIS driver (COM port)

TheGroundZero, 2024-04-11:
If your Raspberry Pi shows up as a COM port instead of a network adapter on your Windows 10 or Windows 11 system, follow the links from this article: https://medium.com/@michi.hammerl/setting-up-a-raspberry-pi-for-usb-gadget-mode-in-windows-11-7993ddf2f80e

Driver: https://github.com/dukelec/mbrush/tree/master/doc/win_driver
[mod-duo-rndis.zip](https://github.com/evilsocket/pwnagotchi/files/14947697/mod-duo-rndis.zip)

[image]

jorgedevs, 2024-08-13:
I tried installing that driver and great, shows up as USB Ethernet/RNDIS Gadget, however, in my Network Connections, it shows up as Network cable unplugged, so I cant ssh into it nor share my internet network connection.

Any guesses? I've looked everywhere but no luck finding a solution.

[image]

elishabr, 2025-05-24:
I have also this problem, and not have solution!!!

bobqianic, 2026-01-18:
> I tried installing that driver and great, shows up as USB Ethernet/RNDIS Gadget, however, in my Network Connections, it shows up as Network cable unplugged, so I cant ssh into it nor share my internet network connection.
> 
> Any guesses? I've looked everywhere but no luck finding a solution.
> 

That works well for me:
https://www.reddit.com/r/raspberry_pi/comments/1i2qggu/raspberry_pi_zero_2w_ethernet_over_usb/

[image]