evilsocket/pwnagotchi #1235: Help with Root access

lordpysc, 2024-06-16:
Ive been trying to install the custom plugins using FileZilla, already did the PermitRootLoging yes and changed the passwd root. But FileZIlla says cannot connect. 
I tried also 10.0.0.2:8080 but when I type Username: root Password: password it asks for it again and again.
Thanks in advance.

[image]
[image]
[image]

timmosaurusrex, 2024-06-16:
Hi there. You need to remove the # before #PermitRootLogin yes. Then you need to restart the ssh service or reboot the system. 
And for the web ui. The default login is username: changeme password: changeme

lordpysc, 2024-06-17:
> Hi there. You need to remove the # before #PermitRootLogin yes. Then you need to restart the ssh service or reboot the system. And for the web ui. The default login is username: changeme password: changeme

Thanks Timmosaurusrex, I'm checking it now. Appreciate your time and support.

lordpysc, 2024-06-17:
Can I bother you with another question? 

If you can help me understand whats going on with the bluetooth tether, i tried with my android phone and while connected to the pc, i was able to pair it, and it connected to my phone. Used the webui 192.168.44.44 and everything worked, but fun thing was after changing ports, from data to power, phone didnt connect and i switch to data and no luck, actually my pwnagotchi disappeared from my phone's Bluetooth devices....

closed 2024-06-18 by lordpysc