evilsocket/pwnagotchi #1256: sh1106_i2c display not supported error

arnav-boop, 2024-09-28:
im trying to use an 0.96inch oled 128x64 i2c display but i get the error 

`unsupported` display type sh1106_i2c
`
i used the instructions from [this](https://www.reddit.com/r/pwnagotchi/comments/14g445o/displays_doesnt_work_need_a_good_guide/) reddit post and i followed ToothNo6591's instructions

**To Reproduce**
Steps to reproduce the behavior:
1. Use aluminium ice's image for pwnagotchi on an raspberry pi 2 zero
2. connect i2c displays vcc to 5v, gnd to gnd. scl to scl and sda to sda.
3. follow instructions from ToothNo6591 on this reddit post 
4. get the error 

**Expected behavior**
I want the ui to be displayed on the oled display and the error `unsupported display type sh1106_i2c` to be not shown/

**Screenshots**
[image]

**Environment (please complete the following information):**
 - Pwnagotchi version - 1.8.5
 - Windows 11
 - Raspberry Pi 2 Zero
 - ssd1306 oled display 0.96 inch

erajtob, 2024-10-03:
The SH1106 and the SSD1306 displays are not exactly identical, for some projects the same code and drivers can be utilized for both displays, but unfortunately the driver for SH1106 that pwnagotchi has doesn't work with the SSD1306 over i2c.

I have ported the SSD1306 display for pwnagotchi and have been using it myself for a while now, if you allow me some time, I can clean the files up and do a pull request sometime within the coming weeks.

[image]

[image]

arnav-boop, 2024-10-03:
Thanks for the reply, after the pull request could you also give a few instructions on how to use the display, I’m pretty new to using the raspberry pi.

Thanks

arnav-boop, 2024-10-11:
Even if the pull request hasn’t been done can you help in making it work without the pull request? @erajtob

erajtob, 2024-10-13:
@arnav-boop I've created the pull request, if you don't wanna wait till the merge, you can always just clone my master branch [erajtob/pwnagotchi](https://github.com/erajtob/pwnagotchi)

If your pwnagotchi is on the latest version as the current one, you can clone my repo to your local machine, I've written a simple bash script that can help you copy over the content to your pwnagotchi over ssh.
- You need to clone my forked repo (master)/ If merged you can just clone the original one here
- You need to have SSH setup for the pwnagotchi [pwnagotchi Docs](https://pwnagotchi.ai/configuration/#connecting-to-pi0w-with-microusb-cable-on-linux-host)
- Copy this script into the cloned pwnagotchi repo directory
  - [install-adafruitssd1306i2c](https://github.com/erajtob/pwnagotchi/blob/d5ca9c47cae21ed9ceca8b6397510e7320564eed/scripts/install-adafruitssd1306i2c.sh)
- Set permission to execute script
  - `chmod +x ./install-adafruitssd1306i2c.sh`
- Execute script
  - `./install-adafruitssd1306i2c.sh`
 
 
The other easy way is of course just clone the repo and manually copy over the [pwnagotchi](https://github.com/erajtob/pwnagotchi/tree/be089667e2a8ca98f31f56575cece9466bbcc3c2/pwnagotchi) directory to your SD card at `/usr/local/lib/python3.7/dist-packages/` that will add the missing files and overwrite the updated files.