evilsocket/pwnagotchi #1259: No LEDs after flashing Pwnagotchi image onto SD card

N00bProgrammer123, 2024-10-13:
No LEDs after flashing Pwnagotchi image onto SD card

I have assembled the hardware of my Pwnagotchi. I'm using a Raspberry Pi Zero 2 W with a WaveShare v2 screen. I have flashed the Pwnagotchi zip file from the lastest release from the official Pwnagotchi website and I'm expecting to show up in Windows etc

I have flashed the image successfully however when I plug it into my PC I have no LEDs on my Raspberry Pi Zero 2 W. There is no image on the WaveShare screen either. I have tried flashing a different OS on and it has worked first time. I have the green light when I use a different OS, however when I use the latest Pwnagotchi img file I have no lights at all.

Could it be the version that of Pwnagotchi that I'm using. I would like to just get it working so if I have to use an older version I will. I mainly want to use Pwnagotchi to learn more about networks etc

timmosaurusrex, 2024-10-13:
The latest image 1.5.5 doesn't support the raspberry pi zero 2. You need to build an image yourself or you can use a fork repo. I recommend aluminum ice fork but you can use different ones. https://github.com/aluminum-ice/pwnagotchi

N00bProgrammer123, 2024-10-13:
Thank you so much I will try it no 👍

N00bProgrammer123, 2024-10-13:
> The latest image 1.5.5 doesn't support the raspberry pi zero 2. You need to build an image yourself or you can use a fork repo. I recommend aluminum ice fork but you can use different ones. https://github.com/aluminum-ice/pwnagotchi

Hello I have tried downloading the aluminum-ice fork and I'm still having no luck with getting the green LED on my raspberry pi after using Balena Etcher when I flash the image onto the SD card. Any other recommendations?

timmosaurusrex, 2024-10-13:
That is weird. You don’t get an error message from balena etcher? Can you maybe connect a screen to the HDMI output of the pi? And when you are done flashing, do you first upload a config.toml or do you boot without the config? Also make sure your power source is strong enough and your sd card is fast enough. Preferably class 10 or higher.

N00bProgrammer123, 2024-10-13:
I have a SanDisk Ultra A1 16GB. I’m getting no errors currently on balena
and I’m sure my power source is good enough as I was able to power it
before with the raspberry pi OS. I’ll try and plug in a HDMI however i do
not have a mouse and keyboard so I’m unsure how this will work

timmosaurusrex, 2024-10-13:
Hmmm. The way to boot raspberry pi os is exactly the same as the way to boot a prebuilt pwnagotchi iso. If your iso is compatible with your hardware, with the aluminum ice fork it is, then it should boot just like normal. I think i cannot help you any further without connecting a screen or something different. Btw, you don’t need a keyboard or mouse. I only use the screen to see if it’s stuck at the rainbow image or some other error or blinking.

N00bProgrammer123, 2024-10-13:
Thank you for all the kind help sir. I will try the screen issue and if it
doesn’t work I’ll probably give up lol. I’ll maybes try another fork but I
have no clue how to use GitHub to find one for the raspberry Pi 0 2.

Thanks dude you’re the best

timmosaurusrex, 2024-10-13:
Never give up lol. I still really have a lot of fun with my 2 pwnagotchi’s. And no problem for helping, I am kinda sad that i am not able to help you. If you have your screen connected, please send the screenshots in here. I am very curious what the screen will show and i hope i can help you with that.

N00bProgrammer123, 2024-10-13:
I have just tried it with a cable connected to my monitor however I just
got the no HDMI signal message on my monitor. I'm assuming because I have
no power LED on it is not powering the HDMI port or the SD card etc. I find
it so strange that it will work with different OS and not a Pwnagotchi. I
will try another YouTube tutorial that is more specific towards the
Raspberry Pi zero 2. I have just checked and the image I have downloaded
from GitHub seems to be corrupt. Any ideas on how I can fix this? This
might be the fix that will get me a working Pwnagotchi.

P.S I know this is a random question but may I ask what you use both your
Pwnagotchi's for? I'm looking forward to using them on my own network and
seeing how they work etc but I was wondering what you use them for. Do you
use them as a hobby or something else. I don't know anyone in person who is
interested in Pwnagotchi's so I find it fun that I'm in contact with
someone who likes them too :)

timmosaurusrex, 2024-10-13:
Yes, that's probably it. To fix the corrupted iso, delete it and download it again. After that, flash it again and hopefully it works. 

To answer your question, I just like to tinker my pwnagotchi's and bring them with me if i go somewhere alone or with my friends. My friends also don't really like that techy gadgets stuff, but i still bring them haha. Around a year after i built my first pwnagotchi, i found a bluetooth gps from 2010 or something new in the box. I was curious if I could get that thing to work with one of my pwnagotchi's and after writing some scripts and tinkering it finally worked. It took me a lot of time but I am still very happy with the result. now I also always bring my GPS to save the coordinates of the handshakes and kinda have my own wardriving map. Maybe I want to make it in the future so that my pwnagotchi also sends the GPS coordinates to my website so that my friends can see where i am walking. They most certainly don't care but i still wanna try if I can do that. I also really recommend to connect you phone to your pwnagotchi with the bluetooth tether plugin. Plugins are also amazing.
Sorry for the long message lol.

N00bProgrammer123, 2024-10-13:
Thank you for the long message. I really appreciated it. Currently I am
downloading the version before just to make sure it is not the current one
that is not working. I realised that the latest version is not a Winwar
file format but the version beforehand is so hopefully that makes
a difference.

I think that is very clever of you to link your Pwnagotchi to your phone
and website. Do you happen to share your website with strangers? I'm
curious to see it. I'm wanting to try take some inspiration if I'm being
truly honest lol

timmosaurusrex, 2024-10-16:
Sorry for the late response, but did re-downloading the ISO work?
And no sorry, my website is not open to public.

N00bProgrammer123, 2024-10-16:
Sorry for the late reply. Unfortunately re-downloading the ISO didn't work.
I tried a few older versions and they did not work :(

I'll have a look on the weekend on GitHub for a different fork hopefully as
I really want to get this working without having to buy a raspberry pi 0
original.

Thanks
Charlie